bigtreetech/BTT-E3-RRF #3: btt tft 35 v2 or 3 or btt mini 12864 v1

mitofun, 2021-04-10:
hi how are you? 
actually i am trying to connect tft 35 or mini 12864 v1 to the boart btt e3 rrf- through reprap and no luck((( https://teamgloomy.github.io/Configurator/Start using this one!)

could you please assist me - need to know the pin for RX1 and RT1 on the board? how should i write it on the firmware?
also if i try to do it frough the marlin no luck ((( 
this message i get- buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/stm32f4/pins_BTT_E3_RRF.h:286:6: error: #error "Only CR10_STOCKDISPLAY, ZONESTAR_LCD, ENDER2_STOCKDISPLAY, MKS_MINI_12864, and TFTGLCD_PANEL_(SPI|I2C) are currently supported on the BTT_E3_RRF."
so when you will do support for btt tft 35 v2 and v3 and btt mini 12864 v1& thank you

psycho0170, 2021-04-12:
Hi, i have the same issue. If you use 1 of the 3 firmware archives that are precompiled you can use it but i would like to configure my Marlin and i can't due to the same issue: #error "Only CR10_STOCKDISPLAY, ZONESTAR_LCD, ENDER2_STOCKDISPLAY, MKS_MINI_12864, and TFTGLCD_PANEL_(SPI|I2C) are currently supported on the BTT_E3_RRF." I would like to know what kind of display i should select to compile my firmware. Thank you.

mitofun, 2021-04-12:
> Hi, i have the same issue. If you use 1 of the 3 firmware archives that are precompiled you can use it but i would like to configure my Marlin and i can't due to the same issue: #error "Only [...]

hi you leave the cr 10 - connect your tft 35 or other to the board + exp3 from tft to exp1 to the board - and install the firmware- works fine))))
this is under marlin mode) 
under the rrf  i have managed to log in under the wifi but not to connect the tft (((

mitofun, 2021-04-12:
under the tft mode (marlin firmware) you cannot use the btt tft 35 - so the question to the BIUQ when you will do the upgrade in the firmware so marlin will supports your tfts? thank you

mitofun, 2021-04-12:
wow i have managed to connect the tft in marlin mode) works good!)))

mitofun, 2021-04-12:
> Hi, i have the same issue. If you use 1 of the 3 firmware archives that are precompiled you can use it but i would like to configure my Marlin and i can't due to the same issue: #error "Only [...]

have managed to connect the tft 35 v3 to the board- in marlin mode))) so all you need to do 👍 
1- in marlin https://github.com/bigtreetech/BTT-E3-RRF open the ender 3 - and set up to your printer paraments- the lcd leave as it is!  for nepixels the pin is pb7- compile and upload the firmware to the board)
2- https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware this is firmware for the tft- 
download and safe to the sd - in the folder - copy to sd card.... the bin file (according to you tft) and the config.ini + theme (tft 35 in my case) and upload it to the tft)
if you have questions you can contact me - +79091151215 whatsap viber telegram etc)

psycho0170, 2021-04-12:
Hi, the only way I can have the touch screen function is uploading last firmware to the screen and then uploading to the board 1 of the 3 firmwares.bin that there are precompiled in the BTT-E3-RRF repository. If i try to compile my own firmware with the Marlin from the repository i always have the same issue: if i not change the screen i cant manage the board from the screen and if i change the screen i havevthe message in visual studio that BTT-E3-RRF board dont support it. So, what kind of screen i have to compile in Marlin to make the touch screen function with board?

Kung-Foo-Kamel, 2021-04-14:
Honestly, if you use Reprap Firmware I would take any display connected to this board and unplug it, and drop it in a draw or a spare parts box, its next to useless, you are by far better off using the web gui, anything you do on the display will only mess things up, some of the features and settings just don't work, and if your to quick with it, it will lock up and you need to reboot the machine to make it work again, all its good for is a basic status display, which you can do with a defunct mobile or tablet through the DWC.

I know its not answering your question, but I should have just saved the money, now that I know, I wont make the mistake again, the RRF web interface is awesome, in every way its better than the little touch display, provided the Wireless performance can improved.

psycho0170, 2021-04-15:
Hi, I dont know how to use reprap, i never tried it. I even dont know how to conect the printer by wifi. Could you explain It a bit please. Thank you.

Kung-Foo-Kamel, 2021-04-15:
Are you on reddit? if so hit me up on their, same name, I will create a post on how to get this board up and running with Reprap, including how to get it started with BT-Touch, as it wont work with the default settings found in this repository.

Kung-Foo-Kamel, 2021-04-15:
Here following this link to my little post: https://www.reddit.com/r/BIGTREETECH/comments/mrcxsv/getting_the_new_bttrrfe3_v10v11_up_and_running/

jaysuk, 2021-04-19:
we also have instructions here https://teamgloomy.github.io/skr_rrf_e3_general.html