bigtreetech/BTT-E3-RRF #4: Wifi Performance

Kung-Foo-Kamel, 2021-04-14:
Hi,

I have everything setup properly, but Wifi performance is extremely bad, its like dial up modem speed, when transferring large files it often fails, and after a while the DWC becomes unusable requiring a reboot of the mainboard.

I have checked and I have 75% signal strength, running a Ubiquity AP in the same room, Antenna is outside the metal chassis, I have searched online for any way to alter additional settings on the ESP8266 but I have only found the basics.

I have tested both mainboard and Wifi firmware from this repository and gloomyandy, both suffer from the same issue.

trengtor, 2021-04-14:
Try to minimize power yours acces point/router.

Kung-Foo-Kamel, 2021-04-14:
Do you mean reduce the output power of my wifi router?

I don't think there is anything incorrect with my Access points, I have a very solid and well managed network, the only thing in my network that is not functioning correctly is this ESP8266.

I run 3 Mesh Amplifi HD's by Ubiquity with gigabit cabled backhaul 3 managed gigabit switches and even the cheapest low-end IOT device on my wifi network is 100% stable, so messing with my Access points will only inhibit other devices from operating at peek performance.

Any advice on how to change what channel its using and how to alter is bandwidth? I feel this may be the problem, but I don't know the console commands to view or alter these features, would be nice if there was a wiki for this board outlining some of these basic commands.

Kung-Foo-Kamel, 2021-04-15:
Update: Disabling the CRC Check will allow stable 300KB transfer rates and I can now transfer over 20MB files without issue, but when trying to copy back to PC, the transfer fails again, I assume its applying the CRC checking on the other end now.

So it doesn't look like a baud rate bottle neck, its possibly something to do with the CRC file checking.

trengtor, 2021-04-17:
> Do you mean reduce the output power of my wifi router?

Yes. This will improve the operation of the radio receiving part of the WiFi module. These are the nuances of radio communication (implementation of receivers in inexpensive WiFi), not a protocol stack.

Kung-Foo-Kamel, 2021-04-18:
I cant do that, the Amplifi HD doesn't have this option, and even if it did, I would prefer to keep it maxed out to squash competing wifi networks, I shouldn't have to retard my wifi network when nothing else is having issues, I'd prefer this device to operate correctly.

I run 3 of these HD's so my property has full coverage, I have IOT devices in close and distant proximity to my AP's all perform great, quite a few of them are running ESP based WiFi boards too, the closest one being about 2 meters from one of the AP's.

Something else to add, I have owned hundreds of different WiFi routers and devices over the past 20yrs, and by far, these Amplifi HD's are the best I have ever had the pleasure of using, they will even beat some of the high end WiFi 6 routers in latency, which I have personally tested and measured as I use them for Wireless VR.

I have found settings in DWC that give me 100% reliability now, but transfer speed is about 40kb per sec, if you disable CRC checking this will rise up to 300kb, at the end of the day, speed does not concern me as prints take hours, so I just upload while its printing something else.

closed 2021-04-18 by Kung-Foo-Kamel