bigtreetech/BTT-E3-RRF #6: USB U-Disk Read Error

Kung-Foo-Kamel, 2021-04-14:
I'm running the TFT35 E3 v3 with the latest firmware, I cant seem to get the USB port to read a USB drive, have tried several USB Flash drives, tried all the usual formats FAT32/NTFS/exFAT it detects the drive being inserted, but gives a read error when trying to access it.

Any advice on what I'm doing wrong here?

I kind of need this to work as transfering gcode files over wifi wont work due to issues with the wifi mentioned in another thread, I feel powering down the machine and pulling the SDcard all the time is not really a good option.

jaysuk, 2021-04-19:
what firmware are you running?
If its RRF, it doesn't support external SD card readers or USB drives

Kung-Foo-Kamel, 2021-04-30:
1. Not an issue anymore as I have decided to remove the display as its complete waste of space, time, money and effort.
2. It does support printing from the SDCARD and USB slot, as these are functions built into the display's GUI and can be accessed from the touch menu, Print -> U-Disk or Print -> TFT SD
3. Reading and printing from the SDCARD in the display slot works
4. Any attempt to read a USB drive will fail
5. Printing from an SDCARD through the display slot will usually fail after a few hours
6. If these features not supported, then I suggest removing all reference of them from the ReRap GUI on the TFT35
7. Also Print -> Onboard SD is useless unless all your gcode's are in the main gcode directory, heaven forbid you try and organise your files in sub directory's.

closed 2021-04-30 by Kung-Foo-Kamel