bigtreetech/BTT-E3-RRF #9: PID Autotune of Hotend and Bed doesn't work on E3-RRF and Ender 3 Pro fom TFT35

Wombat37, 2021-04-17:
Selecting PID Autotune of Hotend or Bed is not working under RepRap system. Autotune operation is confirmed but nothing heats up and eventually the system times out. Latest firmware in both boards.

Works OK from DWC using

M303 H1 P1 S220 C8 ;Hotend PID autotune

or

M303 H0 P1 S60 C8 ;Bed PID autotune

jaysuk, 2021-04-19:
I suggest raising this on this TFT github and not here as its a TFT issue.
Also "C" as an argument on tuning is a Marlin thing and not RRF

Wombat37, 2021-04-19:
Thanks jaysuk for the response. I wasn't too sure where to post this issue. The TFT user interface shows the right dialog but somewhere in the system, the command to perform the PID autotune was getting lost. I don't know which board is doing the losing! My money was on the RRF board because it's new.

Regarding the C argument, the following was taken from the RepRap Wiki (https://reprap.org/wiki/PID_Tuning). It's also listed as a valid argument in the list of RepRap g codes at: https://reprap.org/wiki/G-code#M303:_Run_PID_tuning 

M303 E0 S200 C8

This didn't work with the E0 argument ("no heater has been tuned since startup") but did with the H0 and H1

The T0 argument is not mentioned in the RepRap Wiki or in the official list of g codes for RepRap so you can understand my confusion regarding your posting. However, it did seem to work.

As a newcomer to RRF, this has been my biggest challenge - everywhere you look, there appears to be different answers. It's also apparent that people with experience seem to know the right answers - where did they get them from?

I'm not trying to be negative - I'm just trying to understand this system which looks great but not everything is working properly.

jaysuk, 2021-04-20:
Its definitely not getting lost by the firmware. 
The port to STM32 boards is pretty mature now and has been around for about 8 months now, whereas the TFT firmware has been very flaky until the last month or so.

I wouldn't be using the reprap.org page as it seems to be fairly out of date regarding all the arguments to use.
This page is pretty much your bible https://duet3d.dozuki.com/Wiki/Gcode and you can see that the T arguement is listed for RRF versions 3.2 or greater https://duet3d.dozuki.com/Wiki/Gcode#Section_M303_Run_heater_tuning
We're also quite friendly on our discord channel https://discord.gg/uS97Qs7

Wombat37, 2021-04-21:
Hi jaysuk, thanks for the clarification and for taking the time to respond so fully - it really is much appreciated. I'm still finding my feet with this stuff and welcome the advice you gave on where to get accurate and up-to-date information.

I agree with your assessment of the TFT firmware - it is very flaky and I've had many lockups and crashes - often difficult to reproduce precisely.

Should I be trying to report all this stuff in GitHib? (Evaluating instrumentation firmware was part of my job before retiring).

jaysuk, 2021-04-21:
If there's any issues you find with the TFT firmware etc then please do raise it on github.
BTT have only recently been interested in supporting RRF on the TFT's so they need all the help they can get.
If you have any issues with RRF itself, we would prefer you to report them on the discord channel above. Our developer is around most days and it makes it a lot easier to bug hunt etc in real-time rather than back and forth through github.

f1vefour, 2021-04-22:
@jaysuk Yeah but if the bug reappears there will be no information on GitHub to reference if the discussion takes place in discord.

jaysuk, 2021-04-22:
@f1vefour That would be fine if BTT managed the RRF firmware that runs on the board but they don't.
It relies on our port (TeamGloomy).
Hence Discord is the place to go to report any bugs, not here.

f1vefour, 2021-04-23:
I meant on your git not here, for issues related to the firmware not this 
particular instance.

I'm happy it's been ported regardless, the work which has been done is 
amazing and I thank everyone involved.

jaysuk, 2021-04-23:
We also don't use our git for issue tracking. We use Discord.
Much easier to problem solve issues on the spot than play post tennis.
And pretty much any real issue that gets raised is a duet issue anyway.

Kung-Foo-Kamel, 2021-04-30:
The fix is to unplug the TFT35 and throw it in a draw and forget about using it all, just use a web interface and put up with the slow wifi throughput.

Display cant read subfolders when trying to print, printing from a flash drive in the display will fail after a few hours, the ABL features are nerfed and wont function the same as the web interface, if you do too many things on the display it will eventually lock up making you reset the machine.... I can go on, but I find it a complete waste of time and effort, it may be okay in Marlin, but not for RepRap.

Wombat37, 2021-04-30:
What a miserable email!

If it doesn't work, throw it out! What a message!

Did you ever wonder why things often get better?

I assume that you still live in a cave.

Wombat37, 2021-09-20:
Issue moved to 

[https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1856](url)

It's still open but doesn't belong here.

Closing out...

closed 2021-09-20 by Wombat37