bigtreetech/BTT-E3-RRF #10: Schematic for BTT-E3-RRF V1.1?

kingjamez, 2021-04-19:
I'm trying to trouble shoot the fans on my BTT-E3-RRF V1.1 not working and I suspect that the pin mapping is not correct in the firmware. I need the schematic to know which pins to setup for the fans PWM output.

jaysuk, 2021-04-19:
I can confirm that fan0 is PB5 and fan1 is PB6

kingjamez, 2021-04-19:
> I can confirm that fan0 is PB5 and fan1 is PB6

Oh wow, thanks for the response Jay!  Well I guess I can take that with authority then...
I hate to use you as BTT tech support, but while I have you. Is there any reason short of hardware failure on a brand new board that this wouldn't activate my fans and bed? 
From my RRF config:

; Fans
M950 F0 C"fan0" Q1000                          ; create fan 0 on pin fan0 and set its frequency
M106 P0 S1                                               ;set fan 1 value.
M950 F1 C"fan1" Q1000                          ; create fan 1 on pin fan1 and set its frequency
M106 P1 S1                                               ; set fan 1 value.

M308 S0 P"bedtemp" Y"thermistor" T100000 B3950 ; configure sensor 0 as thermistor on pin bedtemp
M950 H0 C"bed" T0                              ; create bed heater output on bed and map it to sensor 0
M301 P21.73 I1.54 D76.55                       ; disable bang-bang mode for the bed heater and set PWM limit
M307 H0 R2 D65 B1 S1.00                        ; disable bang-bang mode for the bed heater and set PWM limit
M140 H0                                        ; map heated bed to heater 0
M143 H0 S150                                   ; set temperature limit for heater 0 to 150C

Fan 1 seems to be stuck on at roughly PWM of 75%, Fan 0 never activates, and my Bed never starts heating, even when running a M303 calibrate routine.

My hotend, perplexingly, works just fine on H1.

-Jim

jaysuk, 2021-04-19:
can you also post your tool line from your config.g file? should be the line with M563
you fan settings do look wrong though.
which fan is which?

kingjamez, 2021-04-19:
> can you also post your tool line from your config.g file? should be the line with M563
> you fan settings do look wrong though.
> which fan is which?

Those settings are the last configuration from my ~4 hours of testing last night... not stock. There are just designed to make the fans turn. I have a Duet on my big delta and would try settings there before bringing them over to RRF on the Ender 3. 

My current tool settings: 
; Tools
M563 P0 D0 H1                                ; define tool 0
G10 P0 X0 Y0 Z0                                ; set tool 0 axis offsets
G10 P0 R0 S0                                   ; set initial tool 0 active and standby temperatures to 0C

I tried it stock from the github and currently have it with no fan defined for Tool 0 ( I deleted F0 from the M563 line). But I only did that after it didn't work. Note that this tool is working very well... it's the only PWM device working :-).

I have Fan0 on the Hotend fan, Fan1 on the tool fan (part cooling). Both fans are tested and work when directly connected to the power supply.

Thank you for the help! I'm pulling my hair out!

jaysuk, 2021-04-19:
this should work for you
```
; Fans
M950 F0 C"fan0" Q500                           	; create fan 0 on pin fan0 and set its frequency
M106 P0 S0 H-1                                 	; set fan 0 value. Thermostatic control is turned off
M950 F1 C"fan1" Q500                         	; create fan 1 on pin out4 and set its frequency
M106 P1 S1 H1 T45                            	; set fan 1 value. Thermostatic control is turned on

; Tools
M563 P0 D0 H1 F0                               	; define tool 0
```

jaysuk, 2021-04-19:
if it doesn't work, just on discord and we'll go through a few things
https://discord.gg/uS97Qs7

kingjamez, 2021-04-19:
Awesome, will do, thank you!

GadgetAngel, 2021-05-21:
I need to see the schematic so that I can hook up a MAX31865 fir PT100 sensor. I need to see which pins on the board have pull-up resistors and which pins are on SPI bus 1 or SPI bus 2 or SPI bus 3. This is the purpose of the schematic to help users attach additional equipment. Please provide the schematic

trengtor, 2021-05-22:
Please upload schematics files. It needs for non-standart printers.

thisiskeithb, 2023-05-03:
Added: https://github.com/bigtreetech/BTT-E3-RRF/blob/master/BTT%20E3%20RRF%20V1.1/Hardware/BTT%20E3%20RRF%20V1.1-SCH.pdf