bigtreetech/BTT-E3-RRF #12: Nice to have better temperature calibration.

Wombat37, 2021-04-25:
I recently changed my Ender 3 V1.1.4 (8-bit) board to a new BTT-E3-RRF (32-bit) board with E3-TFT35 display running under RepRap. 

I've had a few teething problems but all have been mainly addressed and I'm starting to like the new board and the RRF control a lot!

One major problem I experienced was that the print quality with the new board was awful and there were strange tapping noises coming from the hot-end and there were a lot of under-extrusion issues. Further investigation revealed that the hot-end temperature was about 24°C below the 220°C set point!!!

These temperature measurements were taken using thin thermocouple probes pushed through the hot-end and into the nozzle and confirmed on two meters.

Clearly, my system was badly in need of temperature recalibration. The 'official' instructions I read online to calibrate the hot-end thermistor were a bit vague and involved the removal of the thermistor and measuring its resistance when submerged in an ice bath and in boiling water. This is neither practical nor particularly accurate if the hot-end operating temperature is going to be normally in the range 190°C to 250°C.

There must be an easier way of fixing the temperature calibration on a system like mine when the error is as large as 24°C.

I managed to recalibrate my hot-end temperature in about 15 minutes without removing the thermistor from the system using the following process.

1. With the hot-end off and at ambient temperature, enter the code M308 S1 into the terminal on the TFT35 or via the DWC
2. The system will report the current temperature calibration settings for the hot-end. Note these down (you may need to return to them)
3. Successively enter M303 S1 Txxxxxxx with different values for the T parameter around 100,000 until the correct ambient temperature is reported on the TFT or DWC display. Don't adjust this unless there's a significant error.
4. Set the hot-end temperature to a typical operating temperature (e.g. 220°C)
5. Remove the PTFE tube from the hot-end
6. Push a thin thermocouple probe down through the hot-end and into the nozzle
7. Monitor the temperature of the nozzle on the thermometer connected to this thermocouple
8. Successively enter M303 S1 Bxxxx with different values for the B parameter around 4,000 until the hot-end temperature reading on the thermometer is about the same as the setpoint (within 1 or 2 °C). The previous value for the B parameter on my system was 3950 and I had to adjust it to 4325 to eliminate the 24°C error.
9. Note the new values for the T and B parameters and edit the M303 entry in the config.g file in the sys folder on the SD card to reflect these - e.g. M308 S1 P"e0temp" Y"thermistor" T100000 B4325
10. Remove the thermocouple and re-insert the PTFE bowden tubing

Note that this process is not calibrating the temperature of the thermistor, but rather ensuring that the nozzle temperature is reasonably correct - which is of course much more important.

I've used a similar process to calibrate the hot-bed temperature - the probe is taped to the surface of the bed. 

The reason I'm posting this procedure here is that I feel that a new calibration procedure could be incorporated into the system - this is not a bug but a nice-to-have. I was manually adjusting the values for the T and B parameters but this process could be automated by software. 

Anybody else have any thoughts on this?

P.S. This information may be relevant to the poster of Issue #11

bovoro, 2021-08-06:
Hi Wombat37, that just happened to me, I replaced the thermistor and the ceramic cartridge, and above 200C with a thermocouple it throws me -20C, I have a doubt, in step 3, 8 and 9, you write Successively enter M303 S1 Txxxxxxx with different values for the T parameter, didn't you mean M308? I'm also running Reprap. PD. I saw a video "Marlin Thermistor Calibration (not PID tune)" https://www.youtube.com/watch?v=brl1weuD8bs and calibrates the thermistor in marlin, but it is more laborious and I don't know how we could replicate it in reprap firmware

Wombat37, 2021-08-06:
Hi Bovoro - 

Yes you're quite correct. I did mean M308! I'd been using M303 all evening trying to debug the PID tuning on the TFT35 and must have somehow got it fixated. 

After further thought on this topic, now RepRap supports variables and some limited programmability in macros, I think I can put together a macro to perform this sensor temperature calibration by guiding the user stepwise to take temperature measurements with an independent probe and somehow perform the necessary curve fit on the data. Not sure how to enter values into RepRap macros but everything else looks straightforward.

Wombat37, 2021-08-06:
Think I've worked out how to do all this without needing values to be entered. I'll look into it over the next few days.

bovoro, 2021-08-06:
Yes, I'm keeping an eye on https://duet3d.dozuki.com/Wiki/Calibrating_thermistor_and_PT1000_readings , but Semi-automatic calibration doesnt supported by our board , but at the bottom of that wiki , I had the idea of, select 220C on the screen, and with a thermocouple and multimeter, measure the real temperature, and go sending M308 Snn Lvvv for various values of vvv until I find the value that gives a reading close to the expected value that i setup

Wombat37, 2021-08-06:
My idea would be to have the user manually adjust the hotend temperature control until the thermocouple probe temperature inside the nozzle matches a given fixed temperature. The system will now know the temperature error at that setting. This process could be repeated at 1 or 2 more set temperatures so that we have 2 or 3 temperature error readings across the operational range. Now we can curve fit the expected thermistor profile to the adjusted temperature readings and calculate the coefficients to enter in the M308 statement. I think this could be driven from a RepRap macro.

bovoro, 2021-08-06:
thats sounds great!!. That is beyond my knowledge, I will try to investigate the macros as well,So,  this fixing that you raised at the beginning, can it be used to print as a temporary fix? I guess as long as I don't get too far up or down from the temperature I set to the beta value, right?

Wombat37, 2021-08-06:
The Steinhart–Hart equation used to characterize thermistor R vs T profiles has 3 coefficients although the c coefficient is usually set to zero. To calibrate the other two coefficients you will need at least three data points. The way you suggested might work at the temperature you calibrated it at but may give you a big error if you change that temperature.

bovoro, 2021-08-06:
I was just reading about Steinhart and Hart Equation , thats the key!!!!

bovoro, 2021-08-06:
Hi Wombat37, in flashed today Marlin FW, and the difference of temperature above 200C its only -5C is better than -25C with RRF, I could not compile with VSC, because it gives me error after error after error, I used the precompiled one. do you think it is a failure of poor quality thermistors or firmware configuration issue?  in both Marlin and RRF. with stock one thermistor they works great

bovoro, 2021-08-07:
Wombat37, with datasheet of my 100k b3950 , the datasheet has a table of temperature and resistance, with a calculator https://www.thinksrs.com/downloads/programs/Therm%20Calc/NTCCalibrator/NTCcalculator.htm I calculated the coefficient C steinhart, which apparently should be put in M308 S1 P "e0temp" Y "thermistor" T100000 B4725 C7.06e-8 (it is only an example) in the value of C you put what you get in the calculator in the coefficient C, right now I am printing with marlin, if the print is a hit or miss, I'll test my theory with RepRap Firmware.

Wombat37, 2021-08-07:
Hi Bovoro. I'll work on this later this week. I don't think we should be using this site as a forum so we should perhaps limit our postings. I will try to put together a working system that could be adopted or adapted in the RRF builds

Wombat37, 2021-09-20:
Hi Bovoro (and anyone else who may be interested),

I've looked into this matter of calibrating the temperature of a hot end thermistor and believe that I've come up with a novel solution that easy, cheap, accurate, doesn't need an expensive thermometer, will work with any printer and doesn't require any disassembly or modification.  

I'm not sure how to get these files to you.

Perhaps BTT would be interested in picking this idea up as a potential product?

Wombat37, 2021-09-20:
This request is not appropriate to this repository.

I am still working on this but will close this out.

closed 2021-09-20 by Wombat37