bigtreetech/BTT-E3-RRF #17: RRF and IDEX 1.1

Currents132, 2021-05-17:
I recently received the idex upgrade kit and I assembled it on my ender 3 pro and got everything wired up and loaded the new RepRap support files but the new idex 1.0 board isn't lighting up green and no command work and on the main RRF 1.1 board the 3.3v red led is flashing I have tried using the jumper on the idex board and the DuPont connector from the main board to the idex board and both methods don't work.

Currents132, 2021-05-18:
also my thermistor isn't getting a reading so the value given is -99

Thurnock, 2021-07-12:
So with this add on board you need to go into the marlin/src/pins/stm32f4/BTT rrf E3/ and need to turn on the IDEX board on. it will not work until you do. And also make sure the board is powered with its own 12-24Vdc.