bigtreetech/BTT-E3-RRF #19: BTT-E3-RRF V1.1 it possible to use D-B500F laser modul with RepRap?

Zoli-user, 2021-05-28:
[image]

Hello, My question is very simple. How to connect and how to configure the Creality Ender 3 V2 with BTT-E3-RRF V1.1 to use this laser modul.
I use the latest RepRap firmware and BTT-E3-TFT35. Thank You.

Kung-Foo-Kamel, 2021-11-19:
Not sure about your specific laser module, but most have 4 wires, 12v input for powering the actual laser and a 12v PWM control wires to adjust the level of brightness.

You can configure one of the fan headers to run the PWM control wires and feed the laser module directly with power from your powersupply, if you are using a 24v system and the laser module is 12v, then just power the module from a separate 12v power supply and you can still use the fan header to control the PWN signal, but just set the maximum to 50%, that way your only feeding it 12v max.

So the answer is yes, but you will need to customise the config file to suit the tool your using, I'm actually doing this at the moment, you need to have 2 separate SDcard setups, and swap the card when swapping the tool.

I would be cautious about power your laser module directly from the fan header, you must check the total power required for you specific laser module and make sure the fan header can deliver that power, as the fan PWM outputs on this board can be easily fried, I have a couple here that went poopy in their pants from damaged wiring and faulty cheap fans.

jaysuk, 2021-11-19:
No need to have different SD cards.
Just use M505 https://duet3d.dozuki.com/Wiki/M505

Thanks,

Jay

Kung-Foo-Kamel, 2021-11-21:
Thanks for the info Jay, that's brilliant, this will make things much easier.