bigtreetech/BTT-E3-RRF #20: IDEX board and PT100

sllo13, 2021-06-05:
How to I configure reprap to use a PT100 thermistor with the IDEX board
how to I set the thermistor in the config.g file?
Searched everywhere and havent found an answer

Ragpuss, 2022-01-14:
I would like to know the same thing.
Did you manage to find any answers ?
Also should the heater plug into the IDEX board if using a PT100 on the same board ?
It's strange that there's still no info for this !!

chef79, 2022-01-29:
would love some info about this to

Ragpuss, 2022-01-29:
It looks to me like BTT have combined their PT100 daughter board into the IDEX so if thats the case it should be a setting of 20 for the sensor, I am still not sure on the pin.h file if it will automatically push the connection to the IDEX board like it does for the 2nd Z stepper or if the correct pin assignment has to be put somewhere.

chef79, 2022-01-29:
yes there is pt100 chip on the idex,  but what table do one use to get correct readings.
lowered resistance to 1400 and got 23C  heated the sensor to 60c and temp dropped to 16c hehe.

chef79, 2022-01-29:
heat.tempSensePins = {bedtemp,PT100}; 
you got this in board.txt?  i needed to change E0 to PT100 to get it working

Ragpuss, 2022-01-29:
Isn't E0 the hot end heater ? , it would need to be the thermistor to change.
I haven't changed any pin assignments yet as it's not something that I have messed with much in the past and don't wont to blow my printer up, lol.
Did you have it set to 20 when you got those odd readings. If the IDEX is using 3.3v instead of 5v it might need to be 21
I noticed a jumper that's marked as 3.3v with a jumper cap on it, not sure if that's for the PT100 or not.

marcoeli007, 2023-01-04:
Define in the file "pins_BTT_E3_RRF.h" TEMP_0_PIN to FPC8_PIN. Disable PA0
[image]

Set in "Configuration.h" TEMP_SENSOR_0 to 20
[image]