bigtreetech/BTT-E3-RRF #21: Printer reboots when printing from sd card

asipek1, 2021-06-11:
When printing from sd card printer (Ender 3) reboots after few layers, but when using pronterface that issue is not present. Firmware is Marlin bugfix 2.0.7.2.x., board bigtreetech e3 rrf v1.1.

bovoro, 2021-08-06:
I have the same issue, do you have the web gui configured? its like pronterface wireless, but when i send a command doesnt do anything

asipek1, 2021-08-07:
No I dont have, I resolved my issue by installing new firmware.

randomailer, 2023-04-03:
Hey, I have the same problem. Which firmware did you install?

asipek1, 2023-04-03:
Depends what ender 3 you have, I installed this https://github.com/asipek1/Ender3_BTT_E3_RRF_Marlin/tree/main/3dtouch
Mine is with stock display, 3D touch sensor and e3 rrf motherboard.

randomailer, 2023-04-11:
Thanks, 

It fixed my problem