bigtreetech/BTT-E3-RRF #25: Marlin out dated, to many errors

kastor159, 2021-07-13:
Can the Marlin FW please be updated. 
I find eeprom saves are forgotten.
Z axis offset keeps changing as well.
bad BLtouch functionality.
I fi can install an alternative FW its fine as well.
Thanks

BobCraig10, 2021-08-06:
> Can the Marlin FW please be updated.
> I find eeprom saves are forgotten.
> Z axis offset keeps changing as well.
> bad BLtouch functionality.
> I fi can install an alternative FW its fine as well.
> Thanks

I have nothing to do with this besides owning one also. I planned on using it with reprap firmware, and really liked using it. I did find that using it with Marlin was just a little more "polished" and had some functionality that was more important to me. I was able to compile the newest Marlin 2.0.9x for the btt e3 rrf with no issues. The instructions  to compile are pretty much the same as what is listed to compile for 2.0.7x, just with different line numbers.   Something to look for, the newest Marlin build has provisions to assign 3 serial ports, so you can have the serial tft display, the USB , AND the wifi, so you can have ESP3D functionality  and tft display, and octoprint all work at the same time.  If you need help compiling it, I can try to help, but there may be others that know more about it than me.   If you go through all the Marlin options, there is some really neat features I knew nothing about, and glad I compiled everything myself.

trengtor, 2021-09-01:
> Can the Marlin FW please be updated.

https://github.com/MarlinFirmware/Marlin/issues

nevsdea2, 2022-01-04:
I've had the same problems as @kastor159 - forgotten saves and Z axis offset changes every time I do a print (I need to add the z-axis offset every time I start a print - 2 on the first print, then 4 on the second print, then 6 on the third print, then 8, and so on every time I do a print.

I noticed that 2.0.7.x was a bit old so also wanted to make a new build based off Marlin 2.0.9.2, hoping that using the newer Marlin might help.  I merge in the changes from configuration.h and configuration_adv.h but then it won't build.  It seems I need to add the BIGTREE_E3_RRF environment in the platformio.ini file, then it still won't build because it's missing a bunch of .json and .py files for the BTT and STM32F boards and other stuff.  So I copy those over from the BTT supplied 2.0.7.x build, and then it builds but the board won't boot up when I flash with this firmware.  I can switch back to the 2.0.7.x Ender3 Marlin build and it boots again, but back to the bad behaviours.  

@BobCraig10 how did you get plain vanilla Marlin 2.0.9.x to work?  Maybe you could share your configuration files?  Did you need to modify platformio.ini?