bigtreetech/BTT-E3-RRF #27: What's the best config option on marlin for BTT tft35v3.0 and BTT E3 RFF v1.1?

obionekenewbie, 2021-11-07:
Hi, i recently bought the BTT E3 RFF v1.1 to try the idex mode and the TFT35 V3.0 touchscreen updated to 0.27 but when i compile in VS code i cant choose REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER beacouse if i do i get this error:Only CR10_STOCKDISPLAY, ZONESTAR_LCD, ENDER2_STOCKDISPLAY, MKS_MINI_12864, and TFTGLCD_PANEL_(SPI|I2C) are currently supported on the BTT_E3_RRF. So can you tell me if this board and tft are compatible and in wich way i should set them to work fine? beacouse if i choose CR10_STOCKDISPLAY it "works" only in marlin mode but with a lot of screen bugs and in touch mode the tft says printer not attacked even if try changing the baudrate ( current settings: SERIAL_PORT 3, SERIAL_PORT_2 -1, BAUDRATE 115200.)
Thanks for attention.
Kind regards.

radek8, 2021-11-09:
For touch mode communication, you must connect the serial cable and set serial port 1
SERIAL_PORT 1

[image]

Blacksheep74, 2021-11-17:
OK I've had the same problem, and changing the serial port to 1 still gives me the same error.
Don't understand why this would be released if only compatible with so few LCD displays, and a few searches this is a very big problem with no known solution.

radek8, 2021-11-17:
What's your mistake?

trengtor, 2021-11-17:
[image]
[image]
[image]
[image]

All works fine!