bigtreetech/BTT-E3-RRF #31: Hotend 2 on IDEX build will not settle on the desired temperature

muttstang, 2021-12-30:
I've been fighting this issue for months now on my diy IDEX build.  I have the BTT E3 RRF V1.1 along with the IDEX expansion board.  It oscillates +/- 15 degrees around the desired temperature.  

I have changed out the wires, boards, heaters, thermistors. 

I have run numerous PID tunes and tried manually changing the PID values with no success.  

I have hooked up a multimeter and measured the voltage at the thermistor and it does not track well with what the display shows the temperature to be.  The voltage at the thermistor moves as it should, but the display will show the temperature jumping 10-12 degrees at a single screen refresh.  

I've hooked an oscilloscope up to the thermistors on both hotends and they both give the same amount of noise at the thermistors.  

In looking at the documentation, It appears like the raw thermistor data for the IDEX board is sent over the ribbon cable instead of being converted to a digital signal first so I wonder if it is picking up noise over the ribbon cable.

As I said, I've been trying to debug this issue for a few months and am extremely frustrated.  I can run my printer as a single extruder setup but not as the IDEX as intended.

Here are some links to video I took:

https://youtu.be/qX9mCo87HI8
https://youtu.be/sQQABGbAlfA

jaysuk, 2021-12-30:
Want to come and join us on discord and we can dig into this a bit more? https://discord.gg/uS97Qs7

muttstang, 2021-12-30:
The STM32 hardware support channel?  I'm waiting the obligatory 10 minutes from registration

Thanks
Eric

We must think outside the box! Ya got ta CHANGE the pairadigum! Jesse, step away from the box!

trengtor, 2021-12-31:
Try to use ferrite EMI filter for ribbon cables

Zarrsito, 2023-01-15:
im having the same issue with jumping temps , would be nice if someone finds a working fix for this =)

Zarrsito, 2023-01-19:
[image]
[image]

looks like it gets in trouble when the heatbed gets activated... or disabled, ive tryed manually a bit around ending up with this..
any ideas? When the heatbed gets higher than 60C things normalize except jumping temps on extruder 2, my heatbet makes high tones too, since the upgrade when running on 60C, ive already tryed around with some magnets