bigtreetech/BTT-E3-RRF #34: FAN0 only runs on 100% , 99% wont start it on Marlin / Ender 3

Zarrsito, 2022-07-09:
Hello,

i dont know why but i cant control the Speed of the FAN0 for part Cooling, it only starts on 100% , 99% just make it stop...
on my old creality mainboard there was no problem, since i installed an fan with more power i only run it at 80% cause 100% makes noise :D well the Axis are now Quiet but my fan is noisy now xD any suggestions ??

trengtor, 2022-07-10:
It may depends from cooler model & PWM freq.

> i dont know why but i cant control the Speed of the FAN0 for part Cooling, it only starts on 100% , 99% just make it stop... on my old creality mainboard there was no problem, since i installed an [...]

Zarrsito, 2022-07-10:
Yeah u was right =) i needed to set it to 44000 to function properly, runs smooth now like a charm =)